wake-0/fhvOS #2: [HAL] Add LED functionality to HAL

trylimits, 2015-03-11:
The implemented LED stuff should be refactored to HAL

trylimits, 2015-03-12:
I pushed a prototype implementation how HAL could look like. 
I introduced a platform.h, resp. platform.c which wraps the memory access in inline functions. I'm not entirely happy with the interfaces but I really would like to get rid of these `HWREG(..)` defines in some way.

Let's discuss the concept for that in the coaching tomorrow.

trylimits, 2015-03-15:
Until Wednesday pls @trylimits

trylimits, 2015-03-16:
Implemented but not tested yet. @Blackjack92 I would appreciate if you would review the concept of `API Pin#` and `BeagleBone Pin#`, resp. the mapping between those in gpio.c. Probably we could extend the mapping array and also add all those addresses. This way we could get rid of these `getXXX(..)` functions.
Discussion on Wed.

Blackjack92, 2015-03-16:
I reviewed the concept, it seems good to me. The thing which was missing for my understanding was that the API Pin#s were not in the .h - file defined. When you add them to the .h - file it is okay.

trylimits, 2015-03-17:
API Pin# are wrapped in functions, e.g. `GPIOGetLedPin(int)`. If we would define API Pin#s in the header file we would assume that all hardware devices would have the same amount of GPIO pins.

trylimits, 2015-03-17:
The mux mode in the function `GPIOSetMuxMode(int, int)` is not correct. The mux mode itself should be wrapped into an enum as it may be different on other hw devices.

@trylimits

trylimits, 2015-03-31:
Should work for the moment. Tested with all 4 LEDs

closed 2015-03-31 by trylimits