wake-0/fhvOS #6: [Driver] Implement Driver Manager

trylimits, 2015-03-12:
There should be something like a Driver Manager which handles all the drivers within the kernel.
Rough idea of the concept:

1.) We have x different drivers, each identified by a unique id (of type `uint8`, thus a max of 255 different devices)
2.) DriverManager "instantiates" all drivers and assigns their id
3.) In the kernel itself the drivers can be accessed e.g. like that: `Driver led1 = _drivermanager_getDriver(LED1)` whereby `LED1` is an exported constant by the DriverManager

How user applications, i.e. system API, interconnects with this concept can be discussed later.

trylimits, 2015-03-12:
@Blackjack92 @mpe5651 Please verify the concept and/or feedback.

Blackjack92, 2015-03-12:
sounds good for me

trylimits, 2015-03-31:
Driver Manager is good enough for the moment. Working example

	DeviceManagerInit();

	int i = 0;
	char* leds[] = { "LED0", "LED1", "LED2", "LED3" };
	for (i = 0; i < 4; i++) {
		device_t led = DeviceManagerGetDevice(leds[i], 4);
		DeviceManagerOpen(led);
		DeviceManagerWrite(led, "1", 1);
	}

Adaptions should be treated in another ticket.

closed 2015-03-31 by trylimits