wake-0/fhvOS #11: Get rid of the "test stuff" in main.c

trylimits, 2015-03-16:
We should clean up main.c. Atm there is still prototyping code from UART. This is not acceptable.

It should be cleaned up to:

    int main(void) {
        while(1) { }
    }

@mpe5651 Have you done that yesterday? If so, pls push, otherwise I will take this in the afternoon.

mpe5651, 2015-03-16:
I've already cleaned it up. But I'm not able to push until I get home at approx. 19pm.
Hope you don't need a clean main right away.

Blackjack92, 2015-03-16:
Does anyone need the remaining methods and defines in the main.c? @trylimits @mpe5651

trylimits, 2015-03-17:
See above commit. I moved all the main.c stuff to main_temp.txt in main src folder. main.c is cleaned to endless while-loop.

closed 2015-03-17 by trylimits