wake-0/fhvOS #14: [DeviceManager] Implement a Device Manager

trylimits, 2015-03-17:
The Device Manager should be responsible for managing all devices and their drivers. It should also encapsulate the driver access.

trylimits, 2015-03-17:
@Blackjack92 @mpe5651 This is a first prototype of the Device Manager. Pls review.

Blackjack92, 2015-03-17:
Looks good, very clean. Pls add comments to the functions for a better understanding.

mpe5651, 2015-03-17:
Looks nice to me at the moment. I still have some obscurities about some implementation details, but we'll clarify them tomorrow.

trylimits, 2015-03-31:
Works for the moment. Example of usage:

	DeviceManagerInit();

	int i = 0;
	char* leds[] = { "LED0", "LED1", "LED2", "LED3" };
	for (i = 0; i < 4; i++) {
		device_t led = DeviceManagerGetDevice(leds[i], 4);
		DeviceManagerOpen(led);
		DeviceManagerWrite(led, "1", 1);
	}

Changes and adaptions should be treated in new tickets.

closed 2015-03-31 by trylimits