wake-0/fhvOS #23: [General] Documentation

Blackjack92, 2015-03-28:
Each documentation change, should be added to this "bug".

Blackjack92, 2015-07-08:
Would it be possible to move the existing mmu figures and code snippets to folders in the mmu folder?
So that each module mmu, ipc, processmanagement, can fully be copied by its own without problems.
I will add the folders for the move.
@mpe5651

mpe5651, 2015-07-09:
of course. I have to admit that's a really good idea! I'll take it on at latest by the weekend.

mpe5651, 2015-07-09:
@Blackjack92 @trylimits 

by the way: if you two guys need any UML diagrams, just tell me or send mi a rough hand drawing. I have a commercial licence for enterprise architect (probably the best tool for UML/SysML) and can create diagrams very easily.

Blackjack92, 2015-07-18:
I moved the existing mmu figures and code snippets to folders in the mmu folder. @mpe5651

trylimits, 2015-07-30:
Subsection 2.4 "Zeitplan" refers to an unknown document (MS Project?!). Imo we should either add a screenshot of our schedule or get rid of that subsection. I commented out this subsection and pushed that version for the moment. @Blackjack92 Could you decide pls?

Blackjack92, 2015-07-30:
A screenshot is no good solution for this document (resolution, size, ...). Why should adding a reference to another document be a bad idea? The project file can be attached.

trylimits, 2015-07-30:
From my point of view a documentation should be completely independent of other documents. Either a screenshot or appendix would be my suggestion. If we cannot do that, then my suggestion would be to not go into detail here. We submit the whole package (inc. MS Project file) anyway, so they have access to it if they are interested.

Blackjack92, 2015-07-30:
We need a file which explains what each "file" or "folder" contains. When the documentation is send by mail we have to explain it in the mail itself. So it is no problem for me, when this referencing is done.

trylimits, 2015-07-30:
See new ticket. I will do that in the evening.

trylimits, 2015-07-31:
Finally done :)

closed 2015-07-31 by trylimits