wake-0/fhvOS #27: [UART] [Driver] Write function overflow

trylimits, 2015-04-01:
If we write with full speed into the FIFO queue the queue is overflowed and produces buggy output. We should implement an interrupt which is triggered if the FIFO is empty.

trylimits, 2015-04-07:
If the TX FIFO is full and we keep on writing into the queue the written characters get lost.

There are generally two options to solve this problem:

1.) Implement an interrupt which triggers if the FIFO is empty and refill it again
2.) Write into the FIFO byte-by-byte if and only if the FIFO is not full (wait otherwise)

Above commit implements option 2.

The reason why I've chosen that option is (besides that it is alot easier to implement^^) that it gets a lot more complex if we think of using the UART as console, i.e. delegating `printf(..)` to the uart, e.g. one sends 100 chars to `printf(..)` which usually overflows the TX FIFO. However, the contract of `printf(..)` requires to return to the caller after the characters have been written. Handling the asynchronous interrupt to fulfill this contract correctly seems way too complex :/

mpe5651, 2015-04-07:
If you say it works well than it's fine. 
Otherwise, if we really need them, I can assist you in making the UART interrupts work.

closed 2015-05-01 by trylimits