wake-0/fhvOS #29: [IPC] Implement IPC

Blackjack92, 2015-04-01:
See title.

trylimits, 2015-05-13:
Implemented IPC in the above commits. The implementation is not tested yet. I suggest to review and test together in a coaching.

trylimits, 2015-05-13:
See commit https://github.com/Blackjack92/fhvOS/commit/af300495220aad46734e0a7de6493231404fcd4f

Tested and verified working.

Small code example:

	void ping(void)
	{
		int res = IpcManagerRegisterNamespace("ping");
		
		do {
			res = IpcManagerSendMessage("ping", "pong", "PING");
		} while(res <= 0);
	
		printf("Ping sent...\n");
	
		printf("Ping is ending now...\n");
	}
	
	void pong(void)
	{
		int res = IpcManagerRegisterNamespace("pong");
		while(IpcManagerHasMessage("pong") <= 0)
		{
	
		}
		char message[255];
		char sender[255];
		IpcManagerGetNextMessage("pong", message, 255, sender, 255);
	
		printf("Pong received...\n");
		printf("Message received from \"%s\": %s\n", sender, message);
	
		printf("Pong is ending now...\n");
	}

    int main(void)
    {
        [...]
	SchedulerStartProcess(&ConsoleProcess);
	SchedulerStartProcess(&pong);
	SchedulerStartProcess(&ping);
       [...]
    }

closed 2015-05-13 by trylimits