wake-0/fhvOS #30: [MemoryManagement] Implement MMU

Blackjack92, 2015-04-01:
See title.

mpe5651, 2015-05-07:
By now, coding of the MMU is about 95% complete. I've implemented all major parts and they seem to work (as far as I've been able to test them).
What is left are some minor issus like:
- Switching on the MMU still makes some trouble
- What should happen after a process is killed?
- Test of the dabt-handler
- function for freeing all page frames of a process which is being killed

I would like to ask you two guys ( @Blackjack92  @trylimits ) for a joint review. It shouldn't take more than 2-3 hours to completely make the MMU work.

trylimits, 2015-05-07:
Good work @mpe5651 

@Blackjack92 and I have decided to meet tomorrow for a review session. Maybe we get the MMU to work too, otherwise we should meet on Sat or Sun to finish this.

mpe5651, 2015-05-08:
@trylimits 
ok, very well. write me a mail if you need anything. and please let me know about the results of your meeting :-)

trylimits, 2015-05-17:
Thanks @mpe5651 for your work. I have fixed some minor issues in the code and now the mmu is working (took some time to debug though :P ).

After kicking the scheduler (or process manager) the sequence of dabt interrupts is as follows:

```
CCCCCCCCdabt interrupt from pid=0, with fault state=22
dabt interrupt from pid=0, with fault state=5
Process 1
dabt interrupt from pid=1, with fault state=5
---------Process 2
dabt interrupt from pid=2, with fault state=5
dabt interrupt from pid=3, with fault state=5
led2: i=0
led2: i=1
led2: i=2
led2: i=3
led2: i=4
led2: i=5
led2: i=6
led2: i=7
led2: i=8
led2: i=9
dabt interrupt from pid=4, with fault state=5
process3: i=0
process3: i=1
process3: i=2
process3: i=3
process3: i=4
process3: i=5
process3: i=6
process3: i=7
process3: i=8
process3: i=9
dabt interrupt from pid=5, with fault state=5
[...]
```

The first dabt interrupt (fault state = 22, i.e. 0b10110) corresponds to an asynchronous external abort. I do not find any clear information about the cause of this interrupt (see http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0333h/I1002400.html ), however, not handling this case (as we do) seems to work.

Side note: The asynchronous external abort only happens since the changes made in Bug #63! However, These changes are necessary.

trylimits, 2015-05-17:
Closing this ticket. Further issues regarding the MMU implementation should be managed in new tickets. 

Thanks @mpe5651 for your active and good work.

closed 2015-05-17 by trylimits

mpe5651, 2015-05-17:
thank you for your kind assitance and the time you invested! I'm glad the mmu finally works well :-)