wake-0/fhvOS #31: [SysCalls] Implement System API

Blackjack92, 2015-04-01:
Implement Sytem API e.g. MINIX SysCalls.

trylimits, 2015-05-19:
Some clarifications on the system API:

We should have somewhere a folder `api` which contains all necessary header files and source files for the system calls.
A system call immediately switches into supervisor/privileged mode and executes the corresponding kernel function and switches back to user mode.
The memory mapping of TTBR1 (kernel) has to be read/write-protected for user mode thus triggering a interrupt if a user process tries to access the kernel by bypassing the system API. This ISR should immediately kill the process.

trylimits, 2015-05-19:
Above comment is not entirely true. We discuss the concept in detail at Wed. See #75. If I find some time I will probably implement a prototype for a system call.

trylimits, 2015-05-19:
In the above commit I have refactored some code (according to our conventions) and implemented a working system call from the console to the kernel to start a process.

Discussion tomorrow I guess.

mpe5651, 2015-05-20:
I'm sorry to say that the last commit is not 100% correct. Further explanation at our meeting.

trylimits, 2015-05-23:
I refactored the System API to a more useful structure. The main idea is that all system call functions are wrapped into a user API, such as `execute(char* command)` and within that function a `systemCallMessage_t` is assembled. The user functions should be placed into header files which fulfill the POSIX standard, e.g. `execute(..)` is placed into `system.h`.

Blackjack92, 2015-05-29:
Acc. to todays coaching readfile should only read chunks from the sd card so that there is not the problem that reading needs to much time.

trylimits, 2015-05-29:
Above commits implement the API for accessing the filesystem.

trylimits, 2015-05-30:
The function `SystemCall(..)` in `systemcall.c` includes a switch case handling the different system call ids. However, the `systemCallMessage_t` struct already contains this information. In the kernel implementation we use the system call id from the latter, thus removing this switch case would make the implementation cleaner.

trylimits, 2015-05-30:
Above commit removes this handling and calls the `SWI` instruction with a constant number. All the system calls still work.

trylimits, 2015-07-06:
System API functions are done and are working well. Addtional functions should be discussed in new tickets.

closed 2015-07-06 by trylimits