wake-0/fhvOS #33: [ProcessManagement] Implement Process Manager

mpe5651, 2015-04-01:


Blackjack92, 2015-05-15:
I think this issue can be closed because it was added to ussue #55, which is already closed.
Please verify @mpe5651 @trylimits .

mpe5651, 2015-05-15:
you're right. we oversaw that there already is an open issue for the process manager.

trylimits, 2015-05-19:
Not finished yet :(

trylimits, 2015-05-23:
`ProcessManagerStartProcess` should return the new process id.

trylimits, 2015-05-24:
This `SetProcessState(..)` function should not be provided by the Process Manager. Above commit gets rid of this function.

trylimits, 2015-05-28:
The ProcessManager should provide `argc` and `argv` to the new process.

trylimits, 2015-05-30:
Above commit implements the rest of the functionality of the process manager.

Important note @mpe5651 @Blackjack92: There should not be any interaction with the Scheduler itself regarding creating new and killing processes. For this purpose the API of the ProcessManager has to be used.

closed 2015-05-30 by trylimits