wake-0/fhvOS #44: [Scheduler] Scheduler crashes after some time

trylimits, 2015-05-01:
The scheduler crashes after some time (atm it crashes after one hour). I assume this is caused by not restoring the stack pointer after a context switch, however, we should investigate the problem if it causes major issues.

trylimits, 2015-05-01:
See initial Bug #28 for more information

mpe5651, 2015-05-01:
I think I found the problem. I'll write you later.

trylimits, 2015-05-01:
From Bug #28 @mpe5651 

Take a look at the last line in the restore registers function:

RestoreRegisters:

    ADD SP, SP, #40
    LDMFD SP!, {CPSR}
    LDMFD    SP, {R13}^
    NOP
    ADD      SP, SP, #4
    ;LDMFD   SP!, {SP}
    LDMFD    SP, {R0-R12, PC}^
    ;ADD         SP, SP, #56

We are restoring the registers without correcting the stack pointer. Then the relative slow crash after about an hour would make sense. What do you think?
I'm sorry that I can't test it, I'm currently not st home.

trylimits, 2015-05-01:
I think you are right but the problem is a little bit difficult to solve. The last line restores the context including the PC, thus we cannot do anything below that line. Let's discuss that issue in the next coaching.

mpe5651, 2015-05-02:
Ok. But I would like to invest about 10 minutes tomorrow and test the following sequence:

LDMFD    SP, {R0-R12}^
ADD         SP, SP, #52
LDMFD    SP, {LR}^
ADD         SP, SP, #4
SUBS	PC, LR, #4

trylimits, 2015-05-03:
@mpe5651 Implemented the suggested fix. Further testing needed, but looks good.

trylimits, 2015-05-03:
Commit has not changed the behaviour.

trylimits, 2015-05-17:
Introducing the MMU has fixed this problem. Therefore closing this issue.

closed 2015-05-17 by trylimits