wake-0/fhvOS #48: [Scheduler] Add function for loading from external

Blackjack92, 2015-05-08:
Not sure about this function. 
Add a function, which can be used to load an external process or from an external sd card.

trylimits, 2015-05-12:
Concept should be like this:

1.) External program has to be compiled/linked with our own linker file
2.) Loader should copy the defined sections into memory
3.) `SchedulerStartProcess(..)` should be called with the function which is placed at a specific memory address

trylimits, 2015-05-18:
This bug is tightly related to Bug #46. And to be honest I really don't see any difference. The function which "loads from external" is imo exactly what a loader does.

Blackjack92, 2015-05-18:
So we implement a loader and use it in the scheduler to load an external process?

trylimits, 2015-06-22:
This can be closed as the only issue is to implement a loader which is handled in #46

closed 2015-06-22 by trylimits