wake-0/fhvOS #53: [Console] Extend the set of accepted characters

trylimits, 2015-05-12:
The console should accept:

1. blank character
2. backspace
3. `"` character
4. `*` character
5. `|` character
6. and some more...

trylimits, 2015-05-12:
1.) and 3-5.) done

Backspace support has to be implemented by overriding `__backspace(..)` (see. http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0378d/Chdjigab.html )

trylimits, 2015-05-13:
6.) done

trylimits, 2015-05-18:
I found an issue in using the ` ` character. If there is already an input hitting space will return from `scanf`.

trylimits, 2015-05-18:
Above commits solves the `scanf` issue. It wasn't actually a bug in our implementation :) See http://stackoverflow.com/questions/1247989/how-do-you-allow-spaces-to-be-entered-using-scanf

Blackjack92, 2015-05-18:
Which characters are missing at the moment?

trylimits, 2015-05-18:
Backspace support for sure (but minor) and some other characters which we should add time after time (if needed). Marking this ticket as pending?

trylimits, 2015-05-19:
The console should abort an empty input directly without making a system call.

trylimits, 2015-05-20:
Above commits extends the set of accepted characters and fixed the bug of an empty input. However, we cannot get backspace support if we keep using `scanf` for the prompt input.

The current implementation can be considered a simple, working console which does not support:

1.) Backspace
2.) Command history (up down keys)
3.) Cursor movement

Overall we should consider to implement an advanced console. I will open a new ticket for this.

closed 2015-05-20 by trylimits