wake-0/fhvOS #55: [Scheduler] [Process Management] Add meta information to processes

trylimits, 2015-05-13:
Either the scheduler or a process manager should manage a process's meta information, i.e. name, start time, status etc.

@mpe5651 @Blackjack92 Should we add this functionality to the scheduler itself or should we introduce a process manager which is associated with the scheduler?

mpe5651, 2015-05-13:
I would prefer a process manager. I'll explain you my opinion tomorrow.

Blackjack92, 2015-05-13:
i would also prefer a process manager. It will be the same as a taskmanager.

trylimits, 2015-05-14:
@mpe5651 if i remember correctly this issue should be fixed completely. Could you please close this if I didn't miss something.

mpe5651, 2015-05-14:
you're absolutely right, mr.  @trylimits  :-)

closed 2015-05-14 by mpe5651