wake-0/fhvOS #56: [Driver] SD card

Blackjack92, 2015-05-13:
Implement a SD card driver,which can be used to "communicate" with the SC card.

trylimits, 2015-05-18:
Information and programming model for the multimedia card (MMC) can be found in AM335x technical reference manual on page 3384.

It seems like accessing the SD card is a bunch of work. Discussion on Wednesday. Pending until then.

Blackjack92, 2015-05-24:
Testcode:
'''
	DeviceManagerInit();
	device_t uart = DeviceManagerGetDevice("UART0", 5);
	ConsoleInit(uart);
	printf("Hello \n");

	device_t sdCard = DeviceManagerGetDevice("SDCARD", 6);
	DeviceManagerInitDevice(sdCard);
	DeviceManagerOpen(sdCard);
'''

Blackjack92, 2015-05-27:
@trylimits pls verify the new approach to return the mmcsdCardInfo from the sd card driver read function. So the driver knows nothing about the fat but the filemanager can easily use the existing fat implementation.

trylimits, 2015-05-28:
This bug is related to #74.

trylimits, 2015-06-22:
Seems to work well for the moment. Additional features can be discussed in new (more fine-grained) tickets.

closed 2015-06-22 by trylimits