wake-0/fhvOS #58: [Kernel] Implement kernel functions

mpe5651, 2015-05-14:
The kernel should provide functions like:

1.) `KernelGetUptime()`
2.) ... some more ...

trylimits, 2015-05-17:
@mpe5651 @Blackjack92 I'm not sure about function 2.) Should the kernel really be responsible for managing the current working directory. Shouldn't we introduce some kind of `FileManager` which is responsible for this (and more ofc). Pls provide your oppinions.

trylimits, 2015-05-17:
I have implemented `KernelTick(int)` which counts up the uptime ticks and returns the new value.

trylimits, 2015-05-17:
Above commit implements `KernelInfo`, `KernelError` and `KernelDebug`. Those functions should be used to print to the console within the kernel functions. The interface is equivalent to the `printf` function, therefore something like below is possible:

    KernelInfo("Variable i=%i\n", i);

mpe5651, 2015-05-18:
am I right in the assumption that kernelExecute is for system calls? in that case, implementing this function is done simply by calling a swi with a a appropriate sysCall number.

trylimits, 2015-05-18:
`KernelExecute` is meant to be a function which starts an external program/command, e.g. `KernelExecute("uptime");` starts an external or internal process which prints the uptime. Other examples are `KernelExecute("ls"), `KernelExecute("more <file>") etc. Actually `KernelExecute(..)` just searches for an executable in a specified folder and runs the corresponding program. At the moment we should compile some core processes into the kernel, such as `uptime`.

trylimits, 2015-05-18:
I found an issue with the implemented functions for printing out debug/error/info stuff. These functions use a mutex to ensure ordered output, however, in some cases this weirdly freezes the kernel as most of these functions are called within interrupt handlers which should not be locked :)

trylimits, 2015-05-18:
Above commit solves the mutex issue.

Blackjack92, 2015-05-18:
I would appreciate the FileManager idea. Some instance should handle the current directory with the contained files.

trylimits, 2015-05-19:
The print functions of the Kernel should be wrapped into some conditional statements to allow disabling debug and error output in release mode.

trylimits, 2015-05-20:
Above commit introduces `#define`s in `kernel.c` which allow to control the output level of kernel messages.

trylimits, 2015-05-20:
We should move the `KernelExecute(..)` function completely to the FileManager implementation.

trylimits, 2015-05-28:
This ticket is solved.

closed 2015-05-28 by trylimits