wake-0/fhvOS #67: [Mutex] Implement a mutex

trylimits, 2015-05-17:
We should implement a mutex which block processes from accessing critical sections in parallel.

The implemented mutex has to interact with the scheduler to block the process which accesses a locked mutex. The interface should be as follows:

    MutexLock(&myMutex);     // If myMutex is not locked atm the process may pass, otherwise it should be blocked
    [...] Critical section

    MutexUnlock(&myMutex); // Previously blocked processes should be unblocked, i.e. state ready

trylimits, 2015-05-17:
I have implemented a simple mutex in src/processmanager/mutex.h/c

Usage as suggested in OP.

Test which was used to verify functionality:

```
void simpleProc1(int argc, char** argv)
{
	MutexLock(&mutex);
	volatile int j = 0;
	for (j = 0; j < 10; j++)
	{
		volatile int i;
		printf("Process 1\n");
		for(i = 0; i < 0x0200000; i++);
	}
	MutexUnlock(&mutex);
}
void simpleProc2(int argc, char** argv)
{
	MutexLock(&mutex);
	while(1)
	{
		volatile int i;
		printf("---------Process 2\n");
		for(i = 0; i < 0x0200000; i++);
	}
	MutexUnlock(&mutex);
}

int main()
{
     [...]
     	ProcessManagerStartProcess("sp1", &simpleProc1);
	ProcessManagerStartProcess("sp2", &simpleProc2);
     [...]
}
```

Output:

```
dabt interrupt from pid=0, with fault state=22
dabt interrupt from pid=0, with fault state=5
Process 1
dabt interrupt from pid=1, with fault state=5
Process 1
Process 1
Process 1
Process 1
Process 1
Process 1
Process 1
Process 1
Process 1
---------Process 2
---------Process 2
---------Process 2
[...]
```

trylimits, 2015-05-17:
With above implementation there is still one issue/question open:

What should happen if a process is force killed (e.g. because of NPE) while the process has locked a mutex. Should the mutex be unlocked? Otherwise all blocked processes will be stuck forever. @mpe5651 @Blackjack92

mpe5651, 2015-05-17:
I think in this case we should unlock the mutex. otherwise the blocked processes are in a deadlock. I'll do some research for a possibly better solution.

trylimits, 2015-05-18:
@mpe5651 I agree, but imo this is a little bit hard to implement. If a process gets force killed there has to be an instance which manages all used mutex (or mutexes? :/) and it's owner in order to be able to release the mutex. However, a mutex should be a "standalone" component and should not be registered somewhere else.

Blackjack92, 2015-05-18:
@trylimits  you are right. Which component contains the mutex at the moment?

trylimits, 2015-05-18:
See implementation at `src/processmanager/mutex.c`. For the moment the mutex works properly so shouldn't we mark this as low priority?

Blackjack92, 2015-05-20:
We should implement the suggested change from comment @mpe5651.

trylimits, 2015-05-30:
According to coaching we can assume that the critical section's implementation is bug free. We should not investigate this case further, unless we introduce a ResourceManager in the future, which we probably don't do :)

closed 2015-05-30 by trylimits