wake-0/fhvOS #71: [MMU] NPE should force kill process

trylimits, 2015-05-18:
A process which accesses a NP should be forced killed in the dabt handler.

mpe5651, 2015-05-18:
what exactly do you mean by "accesses a NP"?

trylimits, 2015-05-18:
NP = null pointer :)

I already fixed the issue in the above commit.

Test case was as follows:

```
void nullPointerProcess(int argc, char** argv)
{
	printf("I am accessing a null pointer now \n");

	int* nullPointer = 0;

	printf("Printing the nullpointer: %i\n", *nullPointer);
}
```

I started this process using the `ProcessManager`. Debug output was:

```
[...]
[DEBUG @ 60] dabt interrupt from pid=2, with fault state=5
I am accessing a null pointer now
[ERROR @ 60] NPE from process 2
[DEBUG @ 60] Scheduler is killing process with pid=2
[...]
```

closed 2015-05-18 by trylimits