wake-0/fhvOS #74: [FileManager] Implement FileManager

trylimits, 2015-05-19:
In Bug #58 we discussed the concept of a FileManager which is responsible for managing the current working directory and more. Generally, the FileManager should wrap the access to the file system, therefore opening and reading files should also be done by using the FileManager.

Functions which should be implemented:

1.) `FileManagerOpenExecutable(char* name, boolean_t searchInGlobalBinPath)`
2.) `FileManagerReadFile(char* name)`
3.) `FileManagerGetCWD(..)`
4.) `FileManagerChangeCWD(..)`
5.) `FileManagerReadFolder(..)`

@mpe5651 @Blackjack92 Do you agree with that concept? Suggestions?

trylimits, 2015-05-19:
I have implemented a function which executes a program. Atm. we should also allow hard-coded programs which are compiled into the kernel, see `hardcoded_programs.h` as we do not have access to external storage yet.

Blackjack92, 2015-05-20:
OpenExecutable should provide a parameter to specify if the newly started program should block or fork the old process.

trylimits, 2015-07-06:
Improved File Manager:
1.) Global bin path is searched first for programs
2.) Loader gets the program's name to propagate it to the Process Manager

File Manager DONE.

closed 2015-07-06 by trylimits