wake-0/fhvOS #84: [Kernel] Implement Idle Process

trylimits, 2015-05-21:
Our os needs an Idle process. Otherwise we are not able to kill or block all remaining (non-idle) processes. The idle process should be hardcoded into the os and it should always have the pid=0. We also have to ensure that the process with pid=0 is never blocked and never killed.

One possibility is to add this to the `KernelStart` function or to add it into the scheduler code `SchedulerStart()`.

We also should ensure that the idle process is never running as long as other processes are in `READY` or `RUNNING` state.

mpe5651, 2015-05-21:
please let the PID = 0 unused until further notice. I'll maybe need it to safely change the ASID.

trylimits, 2015-05-21:
Suggestion: Instead of implementing a hard-coded logic in the scheduler avoiding to run the idle-process if other processes are in `READY` state we should implement the idle process as following:

```
while(true)
{
    SystemYield();
}
```

`SystemYield()` should be added to the system API and should immediately switch to the next ready process.

trylimits, 2015-05-24:
The above two commits implement the idle process which follows the concept in the above comment.

Pls close after reviewed.

closed 2015-05-24 by trylimits