wake-0/fhvOS #85: [System API] Extract System API to external library

trylimits, 2015-05-23:
We have to extract the main part of the System API to an external library which can then be linked to external programs.

We should fix this issue in a collective session so each team member has the same project structure within CCS.

Blackjack92, 2015-06-03:
Command to create a bin from an out file:
`
C:\ti\ccsv6\utils\tiobj2bin>tiobj2bin "W:\Fachhochschule\Master\2-term\S1-RessourcenbeschraenkteSysteme\project\fhvOS\applications\helloWorld\Debug\HelloWorld.out" "HelloWorld.bin"
`

trylimits, 2015-07-03:
Extracted system api to external project in the api folder. External programs should add `/Full/FhvOs-API.lib`.

trylimits, 2015-07-03:
Refactored OS implementation accordingly.

Explanation of the API:

The API has to build configurations:

- `Light` which includes all implementations of the API excluding retargeting of the `printf`-implementation to use system calls. This library is linked by the kernel.

- `Full` which includes all source files. External programs should link against this build.

closed 2015-07-03 by trylimits