wake-0/fhvOS #92: [FileManager] Indication of read access through USR1

trylimits, 2015-05-30:
We should implement read access to the MMC by blinking USR1. We should either let the FileManager handle this or the DMA interrupt in the driver.

Blackjack92, 2015-05-30:
Is it possible that another instance as the FileManager makes a read? When this can happen the driver should handle the blink otherwise the FileManager.

trylimits, 2015-05-30:
We should introduce a policy that each access to the file system has to be done through the File Manager. I agree, the FileManager should handle the device. The implementation should follow the standard we set in the Scheduler's implementation, i.e. the `FileManagerInit(..)` function gets a `device_t`.

trylimits, 2015-05-31:
Above commit implements this feature. File Manager now indicates it's work through LED1.

closed 2015-05-31 by trylimits