wake-0/fhvOS #93: [Console] Different colors for outputtypes (info, error, debug)

Blackjack92, 2015-05-30:


trylimits, 2015-05-30:
I agree this would be a nice feature. `KernelInfo` should however print in the default color.

trylimits, 2015-05-30:
Commit looks good overall. However, `ConsoleChangeDisplayedInformationType` should not be public API.

Blackjack92, 2015-05-30:
Refactored the code. Pls have a look @trylimits and close the ticket.

trylimits, 2015-05-30:
I guess you got me wrong :)

The console should not provide API to change the color to debug/error/info. Just make those functions static. The only API the console should provide to public (regarding debug/error/info) are the `KernelDebug/Info/Error(..)` functions.

Or is there any other reason why those functions should be public API?

trylimits, 2015-05-30:
I did some testing with the implemented feature and I found a bug in the implementation:

[image]

Running the `ps` command (which uses `printf` output) uses the debug color for it's output (SBZ = should be white). Imo the color changing is not done correctly. It should be done as follows:

```
... KernelDebug(..)
{
    Change color to green
    [...] Print the debug output
    Reset color to white
}
```

Atm we are changing the color but we are missing to reset it back.

Blackjack92, 2015-05-30:
I do not want to change the color back because it is a lot of overhead. Normally you chose a KernelDebug, KernelInfo or KernelError to display something, so the color should be set always correctly.

Blackjack92, 2015-05-30:
But you are right this change is necessary at the moment. I will commit this change.

trylimits, 2015-05-30:
This statement is true only within the kernel. User processes share the same console but cannot use `Kernel...(..)` functions to output something. User processes have to use `printf(..)` which is not completely under our control. Resetting the color in `Kernel..(..)` functions is therefore absolutely necessary. The overhead of sending 7 characters is acceptable.

Blackjack92, 2015-05-30:
Read commit above.

trylimits, 2015-05-30:
Output color is set correctly now:

[image]

Some minor changes should still be done:

1.) The public API of the console should be `ConsoleColor(color_t)`
2.) Instead of calling `ConsoleDisplayInfo()` for resetting the color we should call `ConsoleColor(white)`

trylimits, 2015-05-30:
As we just discussed:

I agree that `Kernelnfo(..)` should not print white. Blue would be a nice color for `KernelInfo(..)`.

Blackjack92, 2015-05-30:
Maybe this would be a better style for displaying the different stuff (debug, error, info):

[image]

trylimits, 2015-05-30:
I like the output as it is on master branch, i.e. the whole line is colored.

trylimits, 2015-05-30:
@Blackjack92 What purpose has `KernelPrint(..)`? Imo we should refactor this to `KernelVerbose(..)` which makes more sense in the context of printing something system-relevant. I'll open a new ticket for that.

closed 2015-05-30 by trylimits