wake-0/fhvOS #109: [Security] Implement security aspects

mpe5651, 2015-07-17:
reserve first address of every process + implement high vectors

trylimits, 2015-07-27:
Wouldn't it be better to move the start address of the process space higher than the interrupt vectors? This would fulfill the security aspects and it wouldn't be necessary to implement the high vectors.

Blackjack92, 2015-07-27:
It would be a good approach, so that it is not possible to reference the interrupt vectors. Are the high vectors already implemented?

mpe5651, 2015-07-27:
I already moved the starting address of the process space to 0x00024000. now we fulfill the most important security aspects: no possibility of address 0x00000000 allocation and no possibility of exception hacking.
I'll write it down in the documentation together with some alternative solutions.

by the way: high vectors are also implemented. it's up to you two which solution you prefer.

trylimits, 2015-07-27:
What would happen if one accesses the address 0x00010000 for example? Do we get a dabt interrupt? If so don't we have to ensure that the dabt handler kills the process explicitly? Do we already handle this case?

mpe5651, 2015-07-27:
depends on how you are handling the pages in the process space virtual memory region. you can either process them in the dabt handler, or you can just reserve every time a process is generated the addresses from 0x00000000 to 0x00024000. in the second case, a process will be given only pages above 0x00024000. This means you don't need any address checking (e.g. zero address or as you mentioned forbidden addresses) in the dabt handler. 
But one big disadvantage remains: we need do reserve 36 pages for every process, i.e. 36*4kB = 144kB of memory are unused. This is the reason why I would prefer high vectors. Or at least mention this in the documentation. What do you think?

Blackjack92, 2015-07-27:
When it is easy to implement do it, otherwise focus on the documentation.

closed 2015-07-30 by mpe5651